thaliproject/Thali_CordovaPlugin_BtLibrary #9: Working around slow BLE scanning

yaronyg, 2015-10-15:
There is a bug (see [here](https://code.google.com/p/android/issues/detail?id=82463)) in Android that causes BLE scans to take up to 15 seconds. For background work that is fine. But imagine two people pull out the postcard app and expect to start talking immediately. It a'int gonna happen if they have to wait 15 seconds before even discovering each other!

The thread I linked to contains a potential work around which involves replacing one of Android's own C libraries. I honestly don't even know if that is possible but it seems theoretically do able if we are willing to ship our own custom version of the Bluetooth library.

The problem with the proposed solution is that I am worried it might break our discovery mechanism. I just don't understand enough yet about advertising packets in BLE to understand if the Bluetooth address we send is in the advertising packet or is in a SCAN_RSP. If it's in a SCAN_RSP then the suggested work around would actually break us!

yaronyg, 2015-10-15:
Have we experimented with putting the peripheral in different power modes to see how that drives discovery? When the app is active it should probably be high power. But do we actually know if this makes any real difference? Certainly the BLE spec strongly implies it should since it should affect things like advertising intervals.

tompaana, 2016-01-05:
With proper power settings (see AdvertiseMode and ScanMode), this issue goes away. Thus, closing. Reopen if the issue manifests itself again.

closed 2016-01-05 by tompaana