thaliproject/Thali_CordovaPlugin_BtLibrary #11: Bluetooth address format for peer discovery and service discovery appear different

yaronyg, 2015-11-03:
When we look at Logcat and we see devices listed for peer discovery with their Bluetooth address and then we look at the addresses in JSONLINE for service discovery the addresses never match. That is, the addresses we see in JSONLINE are NEVER the ones listed in peer discovery's logcat output. I'm guessing this is because we are somehow serializing the addresses differently. We need this fixed ASAP since it's killing us on debugging.

DrJukka, 2015-11-04:
"peerIdentifier":"08:EC:A9:50:76:27" is the address that is advertised, I would also be using it to connect to the peer over the Bluetooth, and as far as I know its the one that I get when I ask the bluetooth address via the API. 

Where's the JSONLINE and how do we get the addresses shown in there ?

DrJukka, 2015-11-04:
Did quick test with 3 Nexus 5 devices. 
- InstanceString is the string devices create and start advertising.
- Found Service & JsonLine are logs from plugin library level
- peerAvailabilityChanged is log from node.js (callback for ThaliEmitter.events.PEER_AVAILABILITY_CHANGED)

Below is snippeds from logs during the test, all values are matching in all stages, thus I suppose the question is which id used with logs is not matching ?

-------------- DEV3133 ----------------
InstanceString : {"ra":"CC:FA:00:71:F5:C6","pi":"CC:FA:00:71:F5:C6","pn":"DEV3133"}

Found Service, :{"ra":"CC:FA:00:52:DB:FD","pi":"CC:FA:00:52:DB:FD","pn":"DEV1740"}
JsonLine: {"ra":"CC:FA:00:52:DB:FD","pi":"CC:FA:00:52:DB:FD","pn":"DEV1740"} 

Found Service, :{"pi":"CC:FA:00:52:DB:F5","pn":"DEV6202","ra":"CC:FA:00:52:DB:F5"}
JsonLine: {"pi":"CC:FA:00:52:DB:F5","pn":"DEV6202","ra":"CC:FA:00:52:DB:F5"}

peerAvailabilityChanged [{"peerName":"DEV1740","peerIdentifier":"CC:FA:00:52:DB:FD","peerAvailable":true}]
peerAvailabilityChanged [{"peerName":"DEV6202","peerIdentifier":"CC:FA:00:52:DB:F5","peerAvailable":true}]

-------------- DEV1740 ----------------
mInstanceString : {"ra":"CC:FA:00:52:DB:FD","pi":"CC:FA:00:52:DB:FD","pn":"DEV1740"}
 
Found Service, :{"ra":"CC:FA:00:71:F5:C6","pi":"CC:FA:00:71:F5:C6","pn":"DEV3133"}
JsonLine: {"ra":"CC:FA:00:71:F5:C6","pi":"CC:FA:00:71:F5:C6","pn":"DEV3133"}

Found Service, :{"pi":"CC:FA:00:52:DB:F5","pn":"DEV6202","ra":"CC:FA:00:52:DB:F5"}
JsonLine: {"pi":"CC:FA:00:52:DB:F5","pn":"DEV6202","ra":"CC:FA:00:52:DB:F5"}

peerAvailabilityChanged [{"peerName":"DEV3133","peerIdentifier":"CC:FA:00:71:F5:C6","peerAvailable":true}]
peerAvailabilityChanged [{"peerName":"DEV6202","peerIdentifier":"CC:FA:00:52:DB:F5","peerAvailable":true}]

-------------- DEV6202 ----------------

mInstanceString : {"pi":"CC:FA:00:52:DB:F5","pn":"DEV6202","ra":"CC:FA:00:52:DB:F5"}
 
Found Service, :{"ra":"CC:FA:00:52:DB:FD","pi":"CC:FA:00:52:DB:FD","pn":"DEV1740"}
JsonLine: {"ra":"CC:FA:00:52:DB:FD","pi":"CC:FA:00:52:DB:FD","pn":"DEV1740"}

Found Service, :{"ra":"CC:FA:00:71:F5:C6","pi":"CC:FA:00:71:F5:C6","pn":"DEV3133"}
JsonLine: {"ra":"CC:FA:00:71:F5:C6","pi":"CC:FA:00:71:F5:C6","pn":"DEV3133"} 

peerAvailabilityChanged [{"peerIdentifier":"CC:FA:00:71:F5:C6","peerName":"DEV3133","peerAvailable":true}]
peerAvailabilityChanged [{"peerIdentifier":"CC:FA:00:52:DB:FD","peerName":"DEV1740","peerAvailable":true}]
---

yaronyg, 2015-11-04:
This is Yaron's screw up. The peer identifier is, of course, the wi-fi direct address which then leads to service discovery only at which point can we get the bluetooth address which is why they don't match. So what we need to do is to publish in the service discovery the wi-fi address that we did the discovery with so we can match with the peer discovery and know which peer we are connected to.

closed 2015-11-09 by yaronyg