thaliproject/Thali_CordovaPlugin_BtLibrary #12: Fix Incoming Bluetooth connections handshake

DrJukka, 2015-11-05:
with listeningn: https://github.com/thaliproject/Thali_CordovaPlugin_BtLibrary/blob/master/BtConnectorLib/btconnectorlib2/src/main/java/org/thaliproject/p2p/btconnectorlib/BTListenerThread.java

There can be only one incoming handshake active at any given time. This should be changed to handle multiple handshake with array

DrJukka, 2015-11-06:
initial implementation can be found from: https://github.com/thaliproject/Thali_CordovaPlugin_BtLibrary/tree/story_001 needs testing for verifyinf that it works.

DrJukka, 2015-11-10:
Tested the implementation with 5 devices, and Created PR.

yaronyg, 2015-11-10:
Needs to publish this to bin tray and then create a fake pr in thali_cordovaplugin so we can test the fix in CI.

yaronyg, 2015-11-11:
Was blocked on CI not working but has code review.

vjrantal, 2015-11-18:
Done testing locally and in CI and this change does not seem to make anything worse.

The `story_001` branch in the main repository has been updated to use version 0.0.2 from BinTray that already has a version with this commit.

After merging this to `master`, the BinTray version 0.0.2 matches to `master` branch.

closed 2015-11-18 by vjrantal