thaliproject/Thali_CordovaPlugin_BtLibrary #15: Please review BT Plugin Code

yaronyg, 2015-11-10:


tompaana, 2015-11-11:
Some notes of BtConnectorLib review:
- Coding conventions not followed, especially naming needs a lot of work
    - Naming should be both meaningful and consistent
- Java doc missing
- Tags for logging missing
    - private static final String TAG = <class name>.class.getName();
- Use log.e instead of printStackTrace
- Order of things: new data types, constants, members, methods (public, protected, private)
- Favour single points of return instead of multiple
- Redundant local temp variables used where it does not affect thread safety
- I saw magic numbers!
- Is the meaning of State.WaitingStateChange clear enough?
- BluetoothBase, WifiBase - These are not bases for anything
- Architecture: Single public interface would suffice?

closed 2015-11-11 by yaronyg

yaronyg, 2015-11-11:
@tompaana Please file bugs for these issues as they are all technical debt.