thaliproject/Thali_CordovaPlugin_BtLibrary #21: error on connect "read failed, socket might closed or timeout"

joshblour, 2015-12-03:
Hey guys! I've been following along the Thali blog and @DrJukka's articles trying to implement p2p on android. I stumbled across this library and almost cried; it's an answer to my prayers. 

I'm not sure if you guys are ready to accept issues from the public, so if not feel free to shut me down.

I added the library to an empty app to test it out: https://github.com/joshblour/P2pBleTest

the main code is here:
````java
    private void startP2P() {
        if (mConnectionManagerListener == null) {
            mConnectionManagerListener = new ConnectionManager.ConnectionManagerListener() {
                @Override
                public void onConnectionManagerStateChanged(ConnectionManager.ConnectionManagerState connectionManagerState) {Log.d("CML", "connection mgr state changed");}
                @Override
                public void onPeerListChanged(List<PeerDeviceProperties> list) {Log.d("CML", "connection mgr peer list changed: " + list);}

                @Override
                public void onPeerDiscovered(PeerDeviceProperties peerDeviceProperties) {
                    Log.d("CML", "connection mgr peer discovered: " + peerDeviceProperties);
                    mConnectionManager.connect(peerDeviceProperties);
                }
                @Override
                public void onConnected(BluetoothSocket bluetoothSocket, boolean b, String s, String s1, String s2) { Log.d("CML", "connection mgr connected");}
                @Override
                public void onConnectionFailed(String s, String s1, String s2) { Log.d("CML", "connection mgr failed");}
            };
        }

        if (mConnectionManager == null) {
            mConnectionManager = new ConnectionManager(this, mConnectionManagerListener, UUID.randomUUID(), "Yonah1", "_myservice._tcp");
            mConnectionManager.initialize("myPeerID1", "myPeerName1");
            mConnectionManager.start();
        }
    }
````

I've installed the app on several devices (4.4.4, 5.0, & 5.0.2) but haven't been able to successfully establish a connection. When they do find each other on the service callback, there is this error:
````
12-03 13:16:36.388 26427-26427/com.yonahforst.p2pbletest I/org.thaliproject.p2p.btconnectorlib.ConnectionManager: connect: Y635-L01_9517
12-03 13:16:36.389 26427-26427/com.yonahforst.p2pbletest I/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothConnector: connect: Trying to start connecting to null in address F4:E3:FB:01:A1:62
12-03 13:16:36.390 26427-26427/com.yonahforst.p2pbletest I/org.thaliproject.p2p.btconnectorlib.ConnectionManager: onConnecting: null F4:E3:FB:01:A1:62
12-03 13:16:36.390 26427-26427/com.yonahforst.p2pbletest I/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothConnector: connect: Started connecting to null in address F4:E3:FB:01:A1:62
12-03 13:16:36.390 26427-26427/com.yonahforst.p2pbletest D/CML: connection mgr peer list changed: [org.thaliproject.p2p.btconnectorlib.PeerDeviceProperties@32d2dd27]
12-03 13:16:36.390 26427-7082/com.yonahforst.p2pbletest I/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread: Entering thread
12-03 13:16:36.390 26427-7082/com.yonahforst.p2pbletest I/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread: Trying to connect...
12-03 13:16:36.390 26427-7082/com.yonahforst.p2pbletest W/BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback
12-03 13:16:38.977 26427-7082/com.yonahforst.p2pbletest E/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread: Either the socket connection or the construction of a handshake thread failed: read failed, socket might closed or timeout, read ret: -1
12-03 13:16:38.977 26427-7082/com.yonahforst.p2pbletest E/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread: java.io.IOException: read failed, socket might closed or timeout, read ret: -1
12-03 13:16:38.977 26427-7082/com.yonahforst.p2pbletest E/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread:     at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:517)
12-03 13:16:38.977 26427-7082/com.yonahforst.p2pbletest E/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread:     at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:528)
12-03 13:16:38.977 26427-7082/com.yonahforst.p2pbletest E/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread:     at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:320)
12-03 13:16:38.977 26427-7082/com.yonahforst.p2pbletest E/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread:     at org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread.run(BluetoothClientThread.java:84)
12-03 13:16:38.977 26427-7082/com.yonahforst.p2pbletest E/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread: Failed to connect to socket/initiate handshake
12-03 13:16:38.977 26427-7082/com.yonahforst.p2pbletest E/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothConnector: onConnectionFailed: Either the socket connection or the construction of a handshake thread failed: read failed, socket might closed or timeout, read ret: -1
12-03 13:16:38.977 26427-7082/com.yonahforst.p2pbletest I/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread: Exiting thread
12-03 13:16:38.979 26427-26427/com.yonahforst.p2pbletest W/org.thaliproject.p2p.btconnectorlib.ConnectionManager: onConnectionFailed: Either the socket connection or the construction of a handshake thread failed: read failed, socket might closed or timeout, read ret: -1 [myPeerID1 myPeerName1 F4:E3:FB:01:A1:62]
12-03 13:16:38.980 26427-26427/com.yonahforst.p2pbletest I/org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothClientThread: shutdown
12-03 13:16:38.980 26427-26427/com.yonahforst.p2pbletest D/CML: connection mgr failed
````
here are logs from two of the devices: 
API 21 (5.0.2) - https://gist.github.com/joshblour/b244dc7bcca08d7ff4d8
API 19 (4.4.4) - https://gist.github.com/joshblour/791cfb1e0ebefda65592

Another thing I noticed is that often `WifiPeerDiscoverer` will see a device but then `WifiServiceWatcher` will never resolve it.

tompaana, 2015-12-03:
Hiya,

You might wanna try using a fixed UUID in your app (so that it's the same for both peers). Currently, it's random: `mConnectionManager = new ConnectionManager(this, mConnectionManagerListener, UUID.randomUUID(), "Yonah1", "_myservice._tcp");`

Wi-Fi Direct based peer discovery is not reliable and it is a known issue. I'll try to improve it, but I'm also adding BLE based peer discovery, which should provide better results. I actually just started working on that and probably will have something ready next week. See my new branch: https://github.com/thaliproject/Thali_CordovaPlugin_BtLibrary/tree/story_001_tompaana_9

Cheers,
Tomi

joshblour, 2015-12-03:
Hey Tomi! That did the trick! thank you so much.

closed 2015-12-03 by joshblour

tompaana, 2015-12-03:
No worries, thanks for testing! Looks like I have to make sure that the requirement of using a fixed UUID is clearly documented.