thaliproject/Thali_CordovaPlugin_BtLibrary #22: Several connection failures when many devices around

vjrantal, 2015-12-04:
In the CI results, we can see that there are many failures during which something like this is printed to logcat:
```
thali09_LGE-Nexus 5.md:,E/bt-btif ( 2628): DISCOVERY_COMP_EVT slot id:32, failed to find channle,                                       status:1, scn:0
```
In fact, during [one run](https://github.com/ThaliTester/TestResults/tree/51986340441e256_CI_TEST_-_DO_NOT_MERGE_vjrantal/), this was seen quote a few times:
```
$ grep "failed to find channle" * | wc -l
     624
```
The proposal is to try to apply similar workaround than what was discussed at https://bugreports.qt.io/browse/QTBUG-40172.

tompaana, 2015-12-05:
Implemented. Ready for review.

closed 2015-12-08 by yaronyg

vjrantal, 2015-12-08:
@tompaana: Can you paste here a link to CI results where the fixes are in action in a data sending performance test?