thaliproject/Thali_CordovaPlugin_BtLibrary #23: Implement a proper queue for handling BT handshakes

vjrantal, 2015-12-04:
This PR https://github.com/thaliproject/Thali_CordovaPlugin_BtLibrary/pull/14 enabled multiple handshakes to be handled simultaneously. It has been a question mark can this make the overall BT connection process more unreliable in case of many devices.

An improvement that could be tried out would be to implement a proper queue to handle the handshakes.

yaronyg, 2016-09-26:
@evabishchevich / @vasilevskayaem  I've always wondered if this was a problem. Should we handle handshakes serially?

yaronyg, 2016-11-17:
@evabishchevich / @vasilevskayaem I've been waiting 2 months for an answer. Can I please get one?

larryonoff, 2016-11-23:
@evabishchevich / @vasilevskayaem please check the latest @yaronyg comment.

vasilevskayaem, 2016-11-28:
When library starts handshake it's already have bluetooth sockets created. So client just sends massage to server and waits response. A problem here can be only in speed of sending message (because speed when we have several connections become lower), but I don't think we should worry about it in this case because handshakes messages are not long.