thaliproject/Thali_CordovaPlugin_BtLibrary #24: Sending larger amount of data between two devices fails

vjrantal, 2015-12-04:
When I run the data sending performance test locally between two devices and configure the sent data amount to be 10 megs, the test tends to fail. Here https://gist.github.com/vjrantal/1851a2e19b32da86e6da is a clip from the logs that might (or might not) be useful.

What I am wondering is that could this fail, because the other peer starts sending big amount of data that takes longer time and at the same time, the other peer tries to initiate an independent BT connection to the other peer?

Could the BT library "re-use" connections so that if a connection between peers is already available, a single BT connection would be re-used to send data both ways?

yaronyg, 2015-12-04:
I have been wondering if we could re-use connections myself We absolutely will have all of the infrastructure needed to make this happen at the node.js layer since we have to do exactly this for iOS. Also it looks like BlueZ limits connections (at least as of KitKat) to 7 or 8 simultaneously connections (I suspect directly doesn't matter). So if we can re-use connections to send data in both directions that would be a huge win. The only downside is that it cuts our effective data rate in 1/2. It makes me wonder if it might not make more sense to have a traffic cop that makes sure that only one side is trying to replicate at a time. That way if two people are walking by each other we have a higher probability that at least one of them will successfully complete the transfer.

tompaana, 2015-12-05:
This issue should be in [Thali CordovaPlugin project](https://github.com/thaliproject/Thali_CordovaPlugin), not here.

vjrantal, 2015-12-07:
> This issue should be in Thali CordovaPlugin project, not here.

It is technically possible to address the issue either in the native layer (Java in this case) or node layer.

I am currently thinking that due to the "evented-nature" of the node API, it might not be easy to implement this in the node layer in a bullet-proof way and to prevent that never ever BT connection is not tried out to a peer that is already connected. However, I might be wrong and this requires further investigation.

closed 2015-12-15 by tompaana