thaliproject/Thali_CordovaPlugin_BtLibrary #36: Investigate what we can do to survive Marshmallow banning knowing our own MAC

yaronyg, 2016-01-08:
I've written an [article](http://www.goland.org/thalilocalp2psurvivingmarshmallow/) detailing the options we know of. Please start with section 3. I re-wrote the part with details on how to test this out to make it simpler than the email we exchanged and it calls out all the necessary APIs.

closed 2016-02-01 by yaronyg

joshblour, 2016-02-26:
Hey there. I have a friend who is a PM at Google and knows some people on the Android dev team. I've forwarded him your article to pass on. Hope it helps!

yaronyg, 2016-02-26:
@joshblour Thank you! That is awesome. An even better article to send them is http://www.goland.org/fixmarshmallowmac/ where I propose a simple way to allow P2P apps like Thali to function and to achieve the exact same privacy goals as banning the MAC.