thaliproject/Thali_CordovaPlugin_BtLibrary #38: Lower BLE power already when incoming or outgoing connection is tried to be made

vjrantal, 2016-01-13:
Currently, the power is lowered when data is being transferred. This seems to have an implication on making the connection working in the first place.

If BLE power is lowered earlier, maybe the connection succeeds with a better rate.

yaronyg, 2016-01-13:
This argues that we should think really hard about moving beacons over BLE. Otherwise what will happen is:

Time 0 - A phone has BLE on high power and is visible to a bunch of people and can see a bunch of people
Time 1 - The phone sees someone and wants to get their beacons
Time 2 - The phone lowers its BLE power in preparation for pulling in the beacons, now fewer people can see the phone
Time 3 - The phone transfers the beacons over Bluetooth (with BLE still in low power)
Time 4 - The phone finishes the transfer and now raises BLE power and shows up again

If we assume that full data transfers are relatively infrequent (they have to be or we'll kill the battery) then moving beacons over BLE means leaving more battery for moving big data. In other words if we move beacon retrieval to BLE then we only have to set up a bluetooth connection when we know we have serious data to move around.

If we believe http://www.scriptreactor.com/conn_interval-throughput.pdf and we assume we are forced to use the default values then we get about 5 Kb/s. The header for the beacons is 8 bytes for Expiration and 88 bytes for the public key. Each individual beacon adds another 48 bytes. Let's say we have 10 beacons. That means we have (8+88+480)bytes * 8 bits/byte / 1024 bits/Kb = 4.5 Kb. So we should be able to send each set of beacons in about 1 second. Note super great but not horrible.

There is a procedure in BLE called "Read Long Characteristic Value procedure" which is associated with the Read Blob Request/Read Blob Response opcodes. This is actually defined in the [BLE Standard](https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=229737) in Volume 3, Part G, Section 4.8.3. If one looks at [setValue](http://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic.html#setValue%28byte[]%29) in the Android API it accepts a byte array. According to here]([https://code.google.com/p/android/issues/detail?id=158619) it looks like setValue should "just work" with longer values if we have a reasonably modern Bluetooth stack. That article also mentions http://underdark.io which I've seen mentioned in other places as well. They use BLE for meshing. Underdark appears to be using some variant of the Apache license but I'm not 100% sure. Can someone do a diff? Otherwise don't look at them because we don't need license headaches. If they are Apache though then we are good.

So the bottom line is - I think we can move discovery completely to BLE just by reading one characteristic and allow Android to do all the hard work of making sure that the characteristic is cut up into appropriate sized pieces and sent over.

tompaana, 2016-02-10:
This is already done.

closed 2016-02-10 by tompaana