thaliproject/Thali_CordovaPlugin_BtLibrary #40: Would turning off Wi-Fi increase Bluetooth throughput?

yaronyg, 2016-01-22:
If the Thali app is in the foreground or if there are no apps in the foreground and if Thali can detect that Wi-Fi isn't being used (or that we aren't even connected to a Wi-Fi AP) then we might be able to increase Bluetooth bandwidth anywhere from 10x-30x if we turn off Wi-Fi completely just for the transfer period. We should also turn off BLE. This would give us sole control over the 2.4 GHz antenna. An interesting perf experiment to run.

yaronyg, 2016-09-18:
As bug thaliproject/Thali_CordovaPlugin#1000 showed, the answer is yes. I'll open a separate bug to follow up.

closed 2016-09-18 by yaronyg

Vlvic, 2020-06-04:
How can I turn off the BLE on the smartphone without turning off the bluetooth completely?