thaliproject/Thali_CordovaPlugin_BtLibrary #49: Crash with Thread problem BluetoothConnector.java new CountDownTimer

RoundSparrow, 2016-03-06:
If I have an Android device ruing the NativeTestApp and I start/stop the App on other devices this crash happens:

thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothConnector: connect: Failed to create the connection timeout timer: Can't create handler inside thread that has not called Looper.prepare()
   java.lang.RuntimeException: Can't create handler inside thread that has not called Looper.prepare()
	   at android.os.Handler.<init>(Handler.java:200)
	   at android.os.Handler.<init>(Handler.java:114)
	   at android.os.CountDownTimer$1.<init>(CountDownTimer.java:109)
	   at android.os.CountDownTimer.<init>(CountDownTimer.java:109)
	   at org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothConnector$9.<init>(BluetoothConnector.java:605)
	   at org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothConnector.createConnectionTimeoutTimer(BluetoothConnector.java:604)
	   at org.thaliproject.p2p.btconnectorlib.internal.bluetooth.BluetoothConnector.connect(BluetoothConnector.java:348)
	   at org.thaliproject.p2p.btconnectorlib.ConnectionManager.connect(ConnectionManager.java:193)
	   at org.thaliproject.nativetest.app.ConnectionEngine.connect(ConnectionEngine.java:182)
	   at org.thaliproject.nativetest.app.ConnectionEngine.autoConnectIfEnabled(ConnectionEngine.java:549)
	   at org.thaliproject.nativetest.app.ConnectionEngine$2.run(ConnectionEngine.java:500)

RoundSparrow, 2016-03-06:
I've tested this problem another half-dozen times, and I think it may only happen on Android 4.4 device I have.  I know you guys were limiting your version of the App to newer Android versions... so maybe you didn't run into it.  However, I'm not certain if that's the case - as I'm mostly limited to Nexus 5 devices for Android 6.0.1 that only support WiFi discovery (no hardware support for Bluetooth LE on Nexus 5). So different code paths could be involved...

For reference, the problem is a known one: http://stackoverflow.com/questions/32283811/cant-create-handler-inside-thread-that-has-not-called-looper-prepare-in-count

there is already a mHandler using the main looper inside the class that could be used to post the new CountDownTimer() - I'll try it out.

RoundSparrow, 2016-03-07:
I've had good success testing with using the existing mHandler to create the CountDownTimer.  One complexity is that there was one use that immediately started it.  To solve this, I added a boolean param to determine if it should self-start within the Runnable.

Example of changes here: https://github.com/RoundSparrow/Thali_CordovaPlugin_BtLibrary/commit/04d1c791237665a2024d491ca3360763e1112870

tompaana, 2016-03-11:
I think this bug should be fixed in the native test app. The problem occurs only when the app is closing so in my opinion we should make sure auto-connect doesn't happen then.

tompaana, 2016-03-11:
There is now a possible fix. Hope you have time to test it at some point.

closed 2016-03-11 by tompaana