thaliproject/Thali_CordovaPlugin_BtLibrary #50: BluePeerDiscoveryUtils.java int8ArrayToBluetoothAddress leading zero problem results in BluetoothManager.java .getRemoteDevice IllegalArgumentException "is not a valid Bluetooth address"

RoundSparrow, 2016-03-06:
Before the return in the int8ArrayToBluetoothAddress method, try this sanity test:

        if (stringBuilder.toString().length() != "02:00:00:00:00:00".length())
            throw new RuntimeException("This method just sourced a malformed Bluetooth Address!");

RoundSparrow, 2016-03-06:
This method seems to work better, it pads any mac bytes less than 9 with a zero

    private static String int8ArrayToBluetoothAddress(int[] bluetoothAddressAsInt8Array) {
        StringBuilder stringBuilder = new StringBuilder();

        for (int i = 0; i < bluetoothAddressAsInt8Array.length; ++i) {
            // the uppercase "X" on String.format ensures it is uppercase letters, which is convention for a Bluetooth MAC Address
            stringBuilder.append(String.format("%02X", (bluetoothAddressAsInt8Array[i] & 0xff)));

            if (i < bluetoothAddressAsInt8Array.length - 1) {
                stringBuilder.append(BLUETOOTH_ADDRESS_SEPARATOR);
            }
        }

        return stringBuilder.toString();
    }

yaronyg, 2016-03-07:
@RoundSparrow the owner of this is @tompaana and he is heads down on a different assignment through Wednesday, so hopefully he should be able to look at this on Thursday.

tompaana, 2016-03-11:
Many thanks @RoundSparrow! Not only for the interest in the project, but the fix. I'll try to incorporate the fixes to master as soon as possible.

PR #54 (2016-03-16): Fixes and hardening, several issues addressed
* Possible fix to #49: Auto-connect shouldn't now happen when disconnected during closing of the app.
* Fixed #50: Added @RoundSparrow's fix for int8ArrayToBluetoothAddress method.
* Possible fix to #51: Method calling `notifyDataSetChanged` in `PeerListFragment` are now synchronous.
* Fixed #53: If the creation of Bluetooth server socket now fails consecutively 10 times, the connection manager is stopped.
* Updated the BtConnectorLib version number from 0.2.6 to 0.2.7.
* Minor fix: The service data is now printed as hex string in the logs.

---
Th [... cut]
closed 2016-03-11 by tompaana