thaliproject/Thali_CordovaPlugin_BtLibrary #51: NativeTestApp ListAdapter notifyDataSetChanged crashes

RoundSparrow, 2016-03-07:
On devices running Android 5.1 and 6.0 - I'm having random crashes related to threading and notifyDataSetChanged() of the Peer List.  Stress testing the app by stopping and starting multiple peers.

I've had little success so far in tracking down where in the code this is being caused.  I wanted to open an issue in case others wanted to chime in.

tompaana, 2016-03-11:
Did this happen in `PeerListFragment.onDataChanged` or `PeerListFragment.onPeerRemoved`? I wonder if this is due to `notifyDataSetChanged` being called concurrently from different threads.

tompaana, 2016-03-11:
There is now a possible fix. Hope you have time to test it at some point.

yaronyg, 2016-03-14:
We aren't seeing this repro anymore so we are closing. Please re-open if the bug returns. Thanks!

closed 2016-03-14 by yaronyg

RoundSparrow, 2016-03-14:
That's fine. I haven't had time to retest.  I wish I could figure out how I was triggering it - as one day I seemed to be able to do it almost on command ;)