thaliproject/Thali_CordovaPlugin_BtLibrary #58: Connection timeout is not stopping BluetoothClientThread connection attempts

czyzm, 2016-04-27:
When the connection timeout occurs there is a call to BluetoothClientThread.shutdown() method where mIsShuttingDown flag is set. This flag is checked in the BluetoothClientThread.tryToConnect() method after every connection attempt and when it is set no other attempt is made.
The issue is that both methods (tryToConnect and shutdown) are synchronized which means that if we are trying to connect and connection timeout is reached the call to shutdown() waits until we leave tryToConnect() which means that the timeout is ignored. It may lead to further errors if the connection is established as we may get information in listener that connection is established and it is timed out.

tompaana, 2016-04-27:
Removing synchronized keyword from `shutdown` method should be enough, I believe.

czyzm, 2016-04-27:
I suppose two step approach is needed - only setting the mIsShuttingDown flag should be not synchronized. Reseting mListener and calling close() should still be synchronized.

tompaana, 2016-04-27:
But: `BluetoothSocket.close` call should not be synchronized when `BluetoothSocket.connect` is.

tompaana, 2016-04-27:
@czyzm: Could you check if my fix above makes sense?

czyzm, 2016-04-28:
@tompaana: Change looks ok for me

tompaana, 2016-04-28:
@czyzm: That's good and thank you. This was a good find.

PR #59 (2016-05-04): Minor fixes and hardening
* Added possibility to change the manufacturer ID (issue #56)
* Added support for [Thali Cordova Plug-in issue #659](https://github.com/thaliproject/Thali_CordovaPlugin/issues/659)
* Fixed #58

---
This change is [[image]](https://reviewable.io/reviews/thaliproject/thali_cordovaplugin_btlibrary/59)
closed 2016-04-29 by yaronyg