thaliproject/Thali_CordovaPlugin_BtLibrary #63: DiscoveryManager state may be overridden with old value by other thread

czyzm, 2016-05-05:
The DiscoveryManager.updateState() method is called from DiscoveryManager.onBlePeerDiscovererStateChanged() and DiscoveryManager.onWifiPeerDiscovererStateChanged() with current state (mState) as argument as the purpose of those calls is to update the advertising and discovering states, not the DiscoveryManager state. However, the updateState() method is synchronized which may result in situation that some thread (e.g. bluetooth advertiser callback) is calling it with current mState but is blocked. While waiting for execution the mState may change. Later on the blocked thread starts execution but using the previous value of mState overriding the state.

The real life use case:
1. We are starting BLE advertising by calling DiscoveryManager.start().
2. BluetoothLeAdvertiser.startAdvertising is called.
3. Before the DiscoveryManager.start() method sets the RUNNING_BLE state (done at the end) we are getting callback from BluetoothLeAdvertiser (onStartSuccess) which results in other thread issuing DiscoveryManager.onBlePeerDiscovererStateChanged() and finally DiscoveryManager.updateState() with current state (NOT_STARTED) as argument. This call is blocked (synchronized method).
4. DiscoveryManager.start() method sets the RUNNING_BLE state and exits.
5. Previous call to DiscoveryManager.updateState() is continued but with old value NOT_STARTED which results in overriding the state.

yaronyg, 2016-05-05:
Sigh... I knew something like this would happen. The bottom line is that nobody on earth is smart enough to use synchronized. Seriously... it requires having a perfect understanding of all methods and calls everywhere and it's just not possible.

This is a good example. Depending on the vagaries of threading it is trivially easy for calls to be put into unexpected orders and thus give us the wrong result.

At least in the case on onBlePeerDiscoveredStateChanged and onWifiPeerDiscoveredStateChanged we should be able to simplify things by splitting updateState into two methods, updateState(mState) and notifyUpdate(). Since neither of those methods intend to update mState by having them call a dedicated notifyUpdate() we prevent any problem. But this does beg a question that I would like to hear from @tompaana about which is - why do these two methods need to call updateState when neither is trying to change mState? Shouldn't they get notified of an mState change by whomever actually did the update?

For the rest of the methods we have two basic choices.

Synchronize Everything - We currently have 15 different places where updateState is called and all of them can exhibit this race condition. Trying to figure that out is impossible so we could slap synchronize everywhere. My guess is that there will be subtle logic errors in doing this that will cause us to debug for the rest of our lives. So this wouldn't be my first choice. Also this gives a false promise. This would appear to ensure that we always get the correct state but in fact it's actually eventually consistent. The reason is that between the time that a callback is made and synchronize lets it be processed the system state could already have changed.

Centralize the state update logic - There is a potentially simpler approach but it requires a lot more code changes. It follows a couple of rules that make dealing with asynchronous state changes easier.

Rule 1 - Do not pass global values by value - For example, we have a callback, onBluetoothAdapterScanModeChanged which passes in an int which contains the current mode. This is inherently unreliable. The reason is that at Time 0 the mode could change and a call to the callback would be made and at Time 1 the mode could change again and a call to the callback would be made. But depending on how the timing gods are feeling it is perfectly possible for the second callback to get processed before the first! In that case we would always have the wrong value and there would be no recovery!!! Bad. Bad. Bad. So a better approach is that onBluetoothADapterScanModechanged would either not provide the new value or would provide it by reference. Using the example above it wouldn't matter what order the two callbacks came in, the resolution of the reference would always provide the latest value. I would probably use the java.util.concurrent.atomic types for a case like this where we are passing a simple type. Another approach is to not pass any value at all in the callback but expect the receiver of the callback to call a method on the calling class to get the value. In this case that would mean that DiscoveryManager would need a pointer to BluetoothManager and BluetoothManager would need to add a new getMode() class.

Rule 2 - Centralize the state update logic - Right now the logic that goes into calculating mState is distributed across 7 different methods. That is hopeless for anyone to understand. What we need is a single method that knows about all the inputs to calculating mState (e.g. stopWifiPeerDiscovery's updateState, stopBlePeerDiscovered's updateState, bleDiscoveryStarts, wifiDiscoveryStarted, etc. And all of these values would be needed by reference not value. Then whenever one of the callbacks gets a call it can do a little clean up but otherwise it will call the central method that will take a snapshot of the system's various inputs and use that to calculate mState. Of course mState can and will be wrong since even as the snapshot is being taken its possible for values to change. But that kind of eventual consistency is what is required in an asynchronous system.

Note that we could also simplify things by removing WiFi support.

czyzm, 2016-05-06:
We need to take into consideration other "state" elements as well. In particular mShouldBeScanning and mShouldBeAdvertising members. Those are used from not synchronized methods (e.g. onBluetoothAdapterScanModeChanged) when calling synchronized start() method. The same issue may happen resulting, e.g. in restarting discovery after it was stopped.

yaronyg, 2016-05-06:
That's why we need to get the logic for calculating the state into one location so we can reason about it correctly. Note however that because we are truly asynchronous we will be eventually consistent which means that there will be periods of time when the state we advertise will be wrong. But if all the state notification changes get through then eventually we will have the right state and advertise it. This assumes of course that our inputs eventually quies into a consistent state.

czyzm, 2016-05-18:
The fix is delivered within following pull request:
https://github.com/thaliproject/Thali_CordovaPlugin_BtLibrary/pull/57

yaronyg, 2016-05-18:
The general rule we have followed is that once a fix is submitted in a PR (even before the PR is fully accepted) we will close the associated bug. So feel free to close this bug.

closed 2016-05-19 by czyzm