thaliproject/Thali_CordovaPlugin_BtLibrary #70: Instrumentation tests: toggling Bluetooth and WiFi not handling intermediate states

czyzm, 2016-06-10:
The toggleBluetooth and toggleWifi methods used in instrumentation tests are not correctly handling intermediate states (turning on and turning off). They are using "isEnabled" method which is OK for turning on but not for turning off because it assumes the media is turned off even if it is still turning off. We should check the state directly.

yaronyg, 2016-06-10:
I vaguely remember even complaining about this. The real question is, can we fix isEnabled?

czyzm, 2016-06-13:
I treat it as an issue in the instrumentation tests only. This is because we are turning on/off bluetooth and wifi multiple times in a row and not handling the real states results in tries to, e.g. enable bluetooth while it is still turning off.

Of course, the isEnable() method is also used in production code but it is done in a way that is safe (just to check if we can proceed with some operation, not if we can turn it on/off again).

czyzm, 2016-06-13:
Created PR #73

closed 2016-06-13 by czyzm