thaliproject/Thali_CordovaPlugin_BtLibrary #71: The handler for checking scan result is not handling null service uuids.

czyzm, 2016-06-10:
It is possible that ScanRecord.getServiceUuids() is returning null. Our code is not handling this when parsing the scan result in BlePeerDiscoverer.checkScanResult() method.

yaronyg, 2016-06-10:
EEEK! The more interesting question is - why are we getting a null result? I wonder if the code we use to sometimes send a device type ID and sometimes send a service UUID is doing something very wrong. OR if we get a device that is transmitting a device type ID and hence not transmitting a service UUID is causing this. We need to understand which because our current hack of falling back from trying to set the service UUID to setting the device type ID is just wrong.

czyzm, 2016-06-16:
This issue happened once in CI (for about 15 runs) on one device. However, when I took the device to my desk it happened quite often. After debugging it turned out that the null service uuid was used in broadcasting of bluetooth beacons which are stored in the cabinet next to my desk :).
I assume that the issue could also happen just by chance. Nevertheless, as my beacon case shows, we need to be ready for the cases where service UUID is null.

czyzm, 2016-06-20:
PR added: #75

closed 2016-06-20 by czyzm