thaliproject/Thali_CordovaPlugin_BtLibrary #72: Setting discovery mode has inconsistent support checking

czyzm, 2016-06-10:
The DiscoveryManagerSettings.setDiscoveryMode() method is checking for appropriate support when changing the mode. For WIFI it check if WiFi p2p is supported, for BLE it checks if multiple advertisement is supported.

The issue is that this method has no information regarding DiscoveryManager mode - we don't know if we are going to be discovering or advertising or both. If we want to discover we should check for scan batching and filtering suppport and not for multiple advertisement. If we are going to do both we should do all checks. In other words in this method we don't have enough info to decide what should be checked.

My suggestion is to remove check for support from setDiscoveryMode() method. This check should be done (and it is done) when starting advertising/discovering. The setDiscoveryMode() method should just set the required mode and inform listeners about it.

yaronyg, 2016-06-10:
Under the theory of 'do less work' I'm perfectly happy with a situation where BTLibrary fails out if it is on a device that cannot do both discovering and advertising. In theory I realize we could do just one or another but that is actually an optimization.

In other words BTLibrary should pretty much assume that if it is on a device that can't do both discovery and advertisement then it should not be used on that device. Now that having been said if a device has nothing to advertise then it will temporarily turn off advertisement and just turn on discovery in order to save battery.

So the ideal solution would be to have a check when BTLibrary as a whole starts up for "Can this device handle discovery and advertisement" and if no then fail out.

czyzm, 2016-06-13:
Our code right now is splitting those two concepts (discovering and advertising) also in terms of features support. This works well (it is covered pretty well by the tests) so allowing to operate only on devices that support both features would introduce limitation that we can avoid.

I would simply allow setting BLE mode only if BLE is supported but without checking if discovering and advertising is supported. Those checks are done when discovery is started. Such solution makes our code consistent regarding the feature support checks.

czyzm, 2016-06-13:
Let me create PR for that and handle to you.

yaronyg, 2016-06-17:
This seems like a fine solution. Although in practice it means that when an app starts up if it wants to figure out if it can successfully run on the device it will have to actually start discovery and advertising and only then will it learn if there are issues. That seems unfortunate. But let's see if anyone complains. :)

czyzm, 2016-06-20:
PR added: #74

closed 2016-06-20 by czyzm