thaliproject/Thali_CordovaPlugin_BtLibrary #82: BLE Experiments

yaronyg, 2016-07-29:
We really want to move over to BLE for all token exchange but historically we have had issues with the GATT server randomly failing and we have never tested long strings. So this is our experiments in this area.

Right now we have a basic project with a GATT server and a central and we will next teach it to retrieve characteristics in a tight loop to stress it.

yaronyg, 2016-11-01:
The tests we need are:

1. Can we run and connect to a GATT server reliably. In the past we saw time outs where the Gatt server just 'disappeared' and didn't come back for up to 30 seconds. Is that still happening?
2. How reliable is connecting to the GATT server in general?
3. If we give Android a characteristic with a long value (e.g. 4K in our case) and we change that characteristic's value while someone is reading it will this corrupt what the reader gets?

yaronyg, 2016-11-04:
The basic test is to please have the central read a characteristic from the peripheral. The characteristic should be 4K long and the central should confirm that it got the right value. For this test the value should be static. Please run this test in a script in a loop for at least a few minutes. Then put the results of how long it took to connect and pull the data into an array and calculate the median, average and standard deviation for the latency times in the array.

vasilevskayaem, 2016-11-11:
I'm afraid that we could not pass 4k in one characteristic reliable. Probably it's android bug probably just restriction of ble, if I try to read characteristic that is more than 512 bytes, I receive back only first 512 bytes. It doesn't depends on MTU value. If MTU set to 512 bytes app received only one bluetoothGattServer.sendResponse with 512 bytes. If I use default MTU app received 28 bluetoothGattServer.sendResponse with small chunks. 

For samsung S7

MTU | gattConnect speed in ms | read 512 bytes speed in ms
------------ | ------------- | -------------
default 20 | 2612 | 465
512 | 2598 | 451

vasilevskayaem, 2016-11-14:
1) the maximum characteristic size is 512, I checked it in specifications. So we can have 8 characteristics for example to keep 4k data.

2) connection in gatt server is quit reliable. Test app run 2 minutes without errors : it starts scanning, connects to first device, reads characteristic (512 bytes), closes connection, starts scanning again and so on. We should very carefully close all not needed connections, because android can keep only limited amount.

3) We can protect data from corruption. When client starts read characteristic6 server receives android.bluetooth.BluetoothGattServerCallback#onCharacteristicReadRequest(
                BluetoothDevice device, int requestId, int offset, BluetoothGattCharacteristic characteristic) where we decide what we should back in android.bluetooth.BluetoothGattServer#sendResponse(BluetoothDevice device, int requestId,
                                int status, int offset, byte[] value) . So if characteristic was changed server can keep in memory some time old value for particular device or send in sendResponse error status and client will try read characteristic again.

yaronyg, 2016-11-14:
1. Let's test the latency of creating a Bluetooth connection and the speed of reading 4K
2. Let's compare the battery draw of moving 4K over Bluetooth vs BLE

vasilevskayaem, 2016-11-15:
1)I assume that android can't read several characteristic at the same time because in method android.bluetooth.BluetoothGatt#readCharacteristic there is synchronised block. So I've added 8 characteristic and read them one by one. 

2) I also found one more interesting detail in https://www.bluetooth.com/specifications/bluetooth-core-specification/technical-considerations , there are Connection Interval, Slave Latency, Supervision Timeout and in Android we can call gatt.requestConnectionPriority to change those parameters. 

3) I found mistake in calculation for my previous comment, so this comment contains wrong results (I request MTU in wrong place and count connection time from the beginning of scan, not from gattConnect). It looks like connection speed doesn't depends on MTU and Priority.

Speed calculation

MTU | Priority | connection speed in ms | read 4k speed in ms
------------ | ------------- | ------------ | -------------
default ( 20 bytes ) | CONNECTION_PRIORITY_LOW_POWER | 480 | 55179
default ( 20 bytes ) | CONNECTION_PRIORITY_BALANCED | 449 | 20396
default ( 20 bytes ) | CONNECTION_PRIORITY_HIGH | 468 | 5031
512 bytes | CONNECTION_PRIORITY_LOW_POWER | 502| 5307
512 bytes  | CONNECTION_PRIORITY_BALANCED | 493 | 1998
512 bytes  | CONNECTION_PRIORITY_HIGH | 473 | 529

yaronyg, 2016-11-15:
We need to understand how BLE behaves under load. So we should do two fan out tests.

Fan Out 1 - In this test we have one central and 3 peripherals and the central is trying to read from the peripherals.

Fan Out 2 - In this test we have one peripheral and 3 centrals and the 3 centrals are trying to read from the one peripheral.

Fan Out 3 - In this test we have 4 devices, everybody is a central and peripheral and we want to see how long it takes for everybody to get the 4K value from everybody else.

Then we need to run the two tests I discussed above to see how Bluetooth's behavior compares to BLE. We need to compare latency and battery draw.

vasilevskayaem, 2016-11-16:
Ble discharge rate (scan, connect, move 4k date and repeat) is 3.13% 
Bluetooth discharge rate (move 4k date) is 4.51%

Bluetooth read 4k speed is 3-4 ms

Bluetooth create connection speed 223 ms

yaronyg, 2016-11-16:
We are seeing problems when we try to connect to multiple BLE peripherals. We see corrupted messages. Operations that don't finish. Etc. We run into this problem if there is one central and many peripherals as well as the inverse. In fact we are even seeing issues where we have just two devices where each is a central and peripheral and each is trying to read characteristics from each other, we will see read failures even there. It's not good. There was a restriction in the old Samsung phones that they can only handle a single BLE peripheral connection at a time but it's not clear if this applies to new Samsung phones.

yaronyg, 2016-11-23:
Samsung has a problem that if we try to connect to one peripheral from multiple central devices we will see that for a short period of time multiple centrals are connected but quickly centrals will start to be disconnected. Apparently only Bluetooth 4.4 supports simultaneous peripheral and central connections on some chips, not all. But we did see two centrals connect simultaneously but after 1 second the connection was killed, not clear why.

In addition we have found out in general that if Phone A has a central connection to Phone B then Phone B can't have a central connection to Phone A (the case is that both phones are centrals and peripherals). But all that happens is that if both try to connect to each other one will fail.

Experiment 1 - The central immediately requests a characteristic from the peripheral and see if that stops the centrals from getting disconnected

Experiment 2 - We have the central host the gatt server and the peripheral requests the characteristic from the central as soon as the central connects.

vasilevskayaem, 2016-11-25:
today one samsung was not be able create connections at all, after that I thought that probably there is some problems with bluetooth chip. I restarted phone, and I found that phone start works good. I restarted all three phones, and finally I see that two phones can connect to one peripheral (doesn't matter where server and client hosted on peripheral or central). Connection is still unstable, but it works )))

if central immediately requests a characteristic from the peripheral device isn't immediately disconnected and continue works. If app doesn't request characteristic after reconnect sometimes device is get disconnected, but after next connection attempt it works. 

I don't know what cause such errors in samsung that only restart helps fix it.

also app can establish now two ways connection.

vasilevskayaem, 2016-11-28:
It's better to stop scanning while reading or connection. In this case all operation becomes more faster and reliable

vasilevskayaem, 2016-11-28:
with stop scan I've get next speed

phone A-central connected to phone B-peripheral and phone B-central connected to phone A-peripheral. Phone A send 4k message to phone B -- average speed 312ms. Phone A send message to B and B send message to A simultaneously -- average speed for 4k transfer is 348ms

vasilevskayaem, 2016-11-28:
when 3 phones connected to each other ( 6 central - peripheral connections) and only one phone read data -- average reading time for 4k is 978ms

vasilevskayaem, 2016-11-28:
one server and two clients tries read data simultaneously -- average reading time for 4k is 381 ms

one client and two servers, client tries read data from both servers simultaneously -- average reading time for 4k is 498 ms

yaronyg, 2016-11-28:
Test 1 - Run the unit test app, leave it on the phone and then run the BLE native test app

Test 2 - Run the unit test app, shut it down and then run the BLE native test app

yaronyg, 2016-11-28:
Please also figure out if there is an Android API that allows us to statically set a characteristic value on our own GATT server opening the possibility of hardware handoff.

vasilevskayaem, 2016-11-29:
Test 1: BLE test app doesn't work reliable in this case. It's better shut down scanning and advertising in Unit test app. 

Test 2: Ble test app doesn't work reliable, but I didn't see any new test failures in unit test app.

vasilevskayaem, 2016-11-29:
It seems to be that we can't statically set a characteristic value on our own GATT server opening the possibility of hardware handoff. https://developer.android.com/reference/android/bluetooth/BluetoothGattServer.html#sendResponse(android.bluetooth.BluetoothDevice, int, int, int, byte[]) Apps must implement this method when somebody tries to read characteristic.

yaronyg, 2016-11-29:
So now the key question is - can we use BLE and Bluetooth the way we do in Thali and add in the use of the GATT server? Based on test 2 we aren't sure.

vasilevskayaem, 2016-11-30:
there was mistake in my code I didn't close all connections when app was restarted and it feels all cache (kill all connection on server and client side). And it seems that all time when I restart my app something stay in cache and app stops work properly. So I fixed it and retest test 1 and 2. 

So now I see that  only speed slows down (scanning, create connect, read data), also there is limit on bluetooth connections (7 at one time).

closed 2016-11-30 by vasilevskayaem

vasilevskayaem, 2016-12-01:
Tested on Nexus. It works but need some changes in code. Also speed is slower and more reconnections are occurred