thaliproject/Thali_CordovaPlugin_BtLibrary #90: Fix race condition in BlePeerDiscoverer

evabishchevich, 2016-10-17:


evabishchevich, 2016-10-18:
Looks like there is no race condition. I couldn't reproduce the bug from https://drive.google.com/drive/folders/0B3Zint8ZY0O8eFNrWHFRV1IxRVU

yaronyg, 2016-10-18:
We can't make this repro so we have to close for now

evabishchevich, 2016-10-20:
https://gist.githubusercontent.com/evabishchevich/ec887e0589fe336637bbdf7c83f97efa/raw/498d62a843847b60342fd008508cc8646d196182/gistfile1.txt

evabishchevich, 2016-10-20:
s7 - https://gist.github.com/evabishchevich/52a1c2277593c86e5588245b784b7fa9
6p - https://gist.github.com/evabishchevich/52a1c2277593c86e5588245b784b7fa9

evabishchevich, 2016-10-20:
There is a deadlock. We have thread X which calls startScannerAndAdvertiser and gets lock on BlePeerDiscoverer object. Inside startScannerAndAdvertiser thread X calls startAdvertiser. In startAdvertiser it calls setAdvertiseData. Inside setAdvertiseData it calls BleAdvertiser.start() and gets the lock on BleAdvertiser object. At the same time we got callback in main thread that advertisment successfully started, our main thread wants to call setState() in BleAdvertiser but doesn't have a lock, so it waits. Our X thread sets state, finish setAdvertiseData, and release the lock on BleAdvertiser. Main thread wakes up, call setState state, get lock on BleAdvertiser and inside setState calls onIsAdvertiserStartedChanged on BlePeerDiscoverer object. Inside onIsAdvertiserStartedChanged it wants to call synchronized method updateState() but doesn't have a lock, so main thread waits. At the same time thread X tries to call BleAdvertiser.start(), but doesn't have a lock on BleAdvertiser object, so it waits. Our main thread has lock on BleAdvertiser object but wants lock on BlePeerDiscoverer. Our X thread has a lock on BlePeerDiscoverer object but wants lock on BleAdvertiser. It's the deadlock.

evabishchevich, 2016-10-20:
I'm going to provide PR with fix tomorrow morning

evabishchevich, 2016-10-21:
Fixed in evabishchevich_90 branch

commit a2d4f07a (2016-11-03): Fix #90 issue
closed 2016-10-21 by evabishchevich